AxisCommunications/media-stream-library-js #560: Memory leak

yangpu, 2020-12-08:
We are using the library to play live rtsp stream from Hikvison camera, however, the memory of video page is increasing very quick and break the page after several hours.

I have test mjpeg and h264 demos and the memrory of mjpeg page is limited in 100M, but h264 page increasing 1M per second. I guess every video frame is buffering in memory currently.

My question is how to limit the memory or release buffer for living play? Is it a bug or feature to buffer all data? Thanks a lot.

lekoaf, 2020-12-11:
@yangpu does this occur in any specific browser? Or all of them?

I'll ping @steabert to have a look at your question.

yangpu, 2020-12-16:
We have testing this with latest Chrome on MacOS and windows, and Chrome Developer Tools may cause the memory increasing and memory is stable after closing it.

However, the Chrome tab is collapsed for rendering 9 rtsp streams after several hours, and backend tcp-ws-proxy service has consumed 5G memory.

steabert, 2021-02-02:
We don't intend to buffer any data, so there is certainly no intended feature there. We use Node streams for all data, so that any buffering is internal to node. When it comes to rendering live video, anything older than 10s is cleared (see the `mse` component).

If there is a bug somewhere, it would have to be related to keeping a reference to a `msg` somewhere that we shouldn't.

Note that `tcp-ws-proxy` is intended mainly for development, although there should not really be any issues with memory there either (unless the underlying WebSocket server somehow buffers a lot).

fabianeichfeldt, 2021-02-04:
Hey there,

I am experiencing also memory leaks, but my setup is a completely different: I am only using Mp4Muxer and MSE from this repo, and have lot's of other stuff in the frontend.
At first I am investigating if I can reproduce this without my whole frontend stuff.

fabianeichfeldt, 2021-02-12:
I've investigated this a lot, and now I would say: 
MSL works fine, but there is an error in chromium engine: https://bugs.chromium.org/p/chromium/issues/detail?id=1174675&can=2&q=media

fabianeichfeldt, 2021-03-12:
Maybe there is a problem in MSE of Chromium, but at least we can do something as well:
https://github.com/google/shaka-player/issues/2953

I will raise a pull request for this.

steabert, 2021-04-11:
I assigned you to avoid multiple people working on the same thing. If you changed your mind and won't work on it, just unassign yourself.

commit 78aece09 (2021-07-17): fix: revoke object URL to avoid a memory leak (#623)

The problem is discussed in more detail in https://github.com/google/shaka-player/issues/2953

Fixes #560
closed 2021-07-17 by steabert