AxisCommunications/media-stream-library-js #575: how to get frame-by-frrame image from RTP over websocket

JackZhai2018, 2021-01-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Additional context**
Add any other context or screenshots about the feature request here.

how to get frame-by-frrame image from RTP over websocket.
how to design the pipeline to match this need

steabert, 2021-02-01:
You cannot get images from the RTP pipeline, just data. It's the browser that renders the eventual image.

If you use JPEG over RTP, you get the frames as separate images, and they are rendered in a canvas.
When using e.g. H.264, you could take snapshots from the `<video>` element.

Without more details describing your use-case it's difficult to understand your question about the pipeline.

steabert, 2021-04-11:
Closing for now since inactive, but please continue discussion if you have more info, thanks.

closed 2021-04-11 by steabert