AxisCommunications/media-stream-library-js #591: Different video latencies if multiple streams from the same camera are played

fau-xx, 2021-02-08:
We use the axis media stream library extensively to display streams from Axis cameras like the V5915 or P5635-E and we noticed that the latencies differ if the same rtsp url is requested multiple times. If the same is done with the VLC Player all streams are exactly synchronous, which leads me to believe that the same rtsp stream is reused (which is prefered by us). Is there a way to achieve this behaviour using the media stream library?

steabert, 2021-02-23:
When you use RTSP in the browser with media-stream-library, the camera has to send that over WebSocket, which affects the performance, but I'm not sure how much. However, if the camera starts dropping frames because of that, you will definitely see problems with playback, we cannot handle that properly with the MediaSource API, as the browser takes full control over the decoding part.

Which firmware do you use on those cameras?

fau-xx, 2021-02-23:
The AXIS V5915 PTZ Network Camera uses firmware version 8.45.3.1

steabert, 2021-03-31:
If it doesn't occur in VLC it could be related to the WebSocket part, since there the audio and video are multiplexed over TCP. I wonder if you can configure VLC to make it request multiplexed audio/video (I think there is a `mux` option or sth).

closed 2023-03-18 by steabert