AxisCommunications/media-stream-library-js #592: Supported Event  Lists

cubecleveland, 2021-02-08:
HI Guys,

How can i get list of supported events and options available by the library.

the one im specifically interested in is how to replay on disconnect from socket.
I noticed the console spits a warning when the socket or stream is disconnected or stopped i wonder how i can tap into this event.

Roy

jaimish11, 2021-02-25:
Hey man. I was struggling with the same thing. The documentation on methods, events, etc. doesn't exist, unfortunately. 
The following events exist for RTSP: prefinish, end, data, unpipe, error, close, finish. You can tap into any of these by doing something like this:
`const pipeline = new pipelines.CliMp4Pipeline(config)`
`pipeline.rtsp.incoming.on("error", (err)=>{
   //DO SOMETHING HERE
})`

You can check all available options by printing the pipeline object. Since there's no documentation on this stuff, it'll be trial and error for the most part.

fau-xx, 2021-03-05:
I struggle with same problem, but i think, i found the reason why the "onServerClose" event is not called. The event is not propagated unless the Code sent by the server is 1001, which is only returned when: "The endpoint is going away, either because of a server failure or because the browser is navigating away from the page that opened the connection."

```
/**
     * Handler for when WebSocket is CLOSED
     * @param  {CloseEvent} e The event associated with a close
     * @param  {Number} e.code The status code sent by the server
     *   Possible codes are documented here:
     *   https://developer.mozilla.org/en-US/docs/Web/API/CloseEvent
     * @return {undefined}
     */
    socket.onclose = (e) => {
      debug('msl:websocket:close')(`${e.code}`)
      if (e.code === CLOSE_GOING_AWAY) {
        this.onServerClose && this.onServerClose()
      }
      // Terminate the streams.
      incoming.push(null)
      outgoing.end()
    }
```

If you remove the if-condition you should be able to get the event:
```
pipeline.onServerClose  = () => {
   console.log("connection closed");
};
```

steabert, 2021-04-11:
The documentation is mostly on the component level, and in the source itself, so you should get some kind of docs or autocomplete in your editor/IDE when using TypeScript. So, currently you're referred to the source code for documentation. In the example of the `onServerClose`, you're welcome to add an `onSocketClose` that will be called regardless of the code (we have the former callback to distinguish the server closing the socket from any other causes).

There is no real high-level "user" documentation, mostly because there are so many possible use cases. We could generate JSDoc documentation from the source code documentation, but I'm not sure if that would be of great benefit (unless you're not using TypeScript or not relying as much on IDE features).

cubecleveland, 2023-07-14:
@fau-xx do u have a forked version of the library ? how did u end up resolving this ?

closed 2025-01-12 by steabert