AxisCommunications/media-stream-library-js #594: TypeError: BigInt.prototype.toString requires that 'this' be a BigInt

yudataguy, 2021-02-11:
**Describe the bug**
run `yarn lint`, `yarn examples` return error 

```
TypeError: BigInt.prototype.toString requires that 'this' be a BigInt
    at Number.toString (<anonymous>)
    at gotStat (node:fs:1926:19)
    at /Users/*****/Downloads/code/playground/media-stream-library-js/.pnp.js:15471:13
```

**To Reproduce**
`yarn install` - ok
`yarn test` - ok
`yarn lint` - error
`yarn examples` - error

**Environment (please complete the following information):**
 - OS: macOS Catalina
 - Node: 15.8.0
 - Version:latest
 
[image]

steabert, 2021-03-31:
I don't have access to a Mac to reproduce this. Can you check if it works with NodeJS 14 ? (I think we had some problems before with NodeJS 15 and Yarn2).

merceyz, 2021-04-03:
Fixed in https://github.com/yarnpkg/berry/pull/2262, can use `yarn set version from sources` to start using it now before the next release

alainib, 2022-09-23:
i still get this error with cypress.ios 

i have 
yarn 2.4.2 (same error with 2.4.2)
node v16.13.2

i cannot upgrade to yarn 3 and to node v17 because it break the project i'm working on.

my project work good but cypress not :

```
TypeError: BigInt.prototype.toString requires that 'this' be a BigInt
    at Number.toString ()
    at gotStat (node:fs:2650:19)
    at callback (/Users/ai/Library/Caches/Cypress/10.8.0/Cypress.app/Contents/Resources/app/packages/server/node_modules/graceful-fs/polyfills.js:299:20)
    at callback (/Users/ai/Library/Caches/Cypress/10.8.0/Cypress.app/Contents/Resources/app/packages/server/node_modules/graceful-fs/polyfills.js:299:20)
    at /Users/ai/dev/application/.pnp.js:30271:13
```

closed 2025-01-12 by steabert