AxisCommunications/media-stream-library-js #608: Example not working in Chromium

michi7801, 2021-04-11:
**Describe the bug**
The library (or at least the examples) does not working in the Version 92 of the Chromium Browser. After entering the IP in the simple_player example I get the following error:

```
Uncaught DOMException: Failed to execute 'addSourceBuffer' on 'MediaSource': 
The type provided ('video/mp4; codecs="avc1.4d0029"') is unsupported.
    at u.value (http://localhost/axis_test/media-stream-library.min.js:2:250211)
    at MediaSource.a (http://localhost/axis_test/media-stream-library.min.js:2:248988)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Get the Version 92 of the Chromium Browser
2. Download the library, the simple_player.js and the simple.html file from the example section.
3. Set it up on a local webserver
4. See error in console.

**Environment (please complete the following information):**
 - OS: MacOS 11.2.3
 - Browser: Chromium
 - Version: 92.0.4473.0
 - Camera: AXIS P1365 Mk II

steabert, 2021-04-11:
You might be missing H.264 support for your Chromium browser.

michi7801, 2021-04-18:
Thanks, with Chrome it is working. I already knew this works in Firefox, but I was not aware that the Chromium Browser has no h264 Support although Chrome does.

closed 2021-04-18 by michi7801