AxisCommunications/media-stream-library-js #610: Command to Capture Pre-Trigger (buffering) Video

Coppertop12, 2021-04-12:
Hello,

I am trying to modify a custom program to include buffered video from before an event trigger. I know this is possible because the event trigger on an axis camera can record before an event, however I am not sure what command or structure to use. 

I cannot provide example code without permission from the original author, working on that to include in my question.

rikteg, 2021-04-13:
Hi
As far as I am aware, it is not possible to request a stream using that mechanic. That buffer is only internal in the cameras. Or at least I have never heard of anyone doing this by only using the api's.
You could try to just read the documentation, https://www.axis.com/vapix-library/. If there is something you could use, it should be documented there, probably under "Network video > Video streaming"

Coppertop12, 2021-04-15:
Thanks for the comment! Definitely pointing me in the right direction. That page is a wealth of information. I am currently seeking answers to all my questions there. Current questions:

Taking a very new tack here. Any ideas on how to trigger a virtual input (using a batch file right now) Without opening a chrome/explorer window?

Command: http://<servername>/axis-cgi/virtualinput/activate.cgi?<argument>=<value>&<argument>=<value>

I have tried several  -commands for chrome to run minimized, but I think it is not logging into the administrative level of the camera as is necessary to trigger the virtual input.

Or, if there is a simple command I can run from a script to trigger the input using AxAXISMEDIACONTROLLib that would be great.

rikteg, 2021-04-16:
You could have a look at curl: https://curl.haxxs.se. That is a very versatile command-line tool for working with requests

closed 2025-01-12 by steabert