AxisCommunications/media-stream-library-js #665: Broken player when bundling with rollup (Vite).

JanKaifer, 2022-04-12:
**Describe the bug**
Everything works in CRA or [Vite](https://github.com/vitejs/vite) dev-mode (esbuild).
But it breaks when bundled with Vite prod-mode (rollup).

**To Reproduce**
1. Bootstrap Vite project
2. Add `<Player hostname="..."/>`
3. run `yarn build`

I created a simple project here: https://stackblitz.com/edit/vitejs-vite-afydj5?file=src/App.tsx
But you need to force your browser to connect to camera via http from a https site.
Or you can download it and run it locally from `localhost`.

**Screenshot**
[image]

**Environment (please complete the following information):**
 - OS: Linux&Mac
 - Browser (all tested browser had this issue): Mac(Chrome, Firefox), Linux(Chrome, Firefox)
 - Version 7.0.0

**Additional context**
Vite is a popular alternative to CRA.

**Workaround**
We just manually add `<script/>` with a minified build of this library.
Then it works as expected, but it's suboptimal.

steabert, 2023-03-18:
Looking at your example project you're not replacing `streams` with `stream-browserify` ? That's an alias that is in our example vite configs which you'll also need when building.

When you use the bundle that was built with esbuild, it has already done that since esbuild respects the browser field in `package.json` to replace the package.

closed 2023-03-18 by steabert

jankaifer, 2023-03-18:
Thanks a lot for the reply. I don't work that project anymore, but I'll pass this along.