AxisCommunications/media-stream-library-js #691: DOMException: Failed to execute 'appendBuffer' on 'SourceBuffer': The HTMLMediaElement.error attribute is not null.

heppth, 2023-01-23:
**Describe the bug**
The Html5VideoPipeline is working perfectly for most devices. However, the H.264 stream of one device cannot be played. The following error appears in the logs.

```
msl:mse failed to append to SourceBuffer:  +0ms DOMException: Failed to execute 'appendBuffer' on 'SourceBuffer': The HTMLMediaElement.error attribute is not null.

{
  "code" = 3
  "message" = "PIPELINE_ERROR_DECODE: Failed to send audio packet for decoding: {timestamp=0 duration=64000 size=526 is_key_frame=1 encrypted=0}"
}
```
It seems that there is a problem with audio decoding. I don't know if the problem is on the camera side or the player side. In the settings of the camera the sound was disabled. When I open the RTSP URL in VLC, the stream works without problems.

Is there a possibility to not setup an audio track. So to explicitly request only a video track (as a possible work-around)?

**To Reproduce**
My test code is pretty simple:
```
            this.$video = $('<video autoplay muted id="test">').insertAfter(this.$canvas);

            const mediaElement = <HTMLVideoElement>this.$video[0];

            const pipeline = new mediaStreamLibrary.Html5VideoPipeline({
                ws: { uri: `ws://${window.location.hostname}/cameraTest` },
                rtsp: { uri: `rtsp://192.168.20.211:554/h264Preview_01_sub` },
                auth: { username: '******', password: '******' },
                mediaElement,
            })
            pipeline.ready.then(() => {
                pipeline.rtsp.play()
            })
```

**Screenshots**

Logs:
[image]

HTMLMediaElement.error attribute:
[image]

**Environment:**
 - OS: Windows
 - Browser: Chrome
 - Version: 109.0.5414.75

**Additional context**
I attached the complete log of the browsers console: [mediaStreamLibrary_Log.txt](https://github.com/AxisCommunications/media-stream-library-js/files/10478004/mediaStreamLibrary_Log.txt)

steabert, 2023-03-18:
From the logs it appears that your camera is telling you that there is audio included? After that I'm not realy sure, do you see any audio data arriving? You could try to debug the AAC depay component if it's loaded.

steabert, 2023-09-23:
Stale. Please re-open if more info.

closed 2023-09-23 by steabert