AxisCommunications/media-stream-library-js #726: Media stream player CRA React fail

cmpeguerog, 2023-01-25:
**Describe the bug**
Unable to use stream-media-player on a clean CRA with installed peer dependencies. When **yarn start** command is invoked this error is prompted: **Can't resolve 'media-stream-library' in 'xxx\node_modules\media-stream-player\dist'**

**Environment (please complete the following information):**
 - OS:Windows
 - Browser: Chrome

steabert, 2023-03-18:
Thanks for your report. Do you have a minimal example how to reproduce this?

steabert, 2023-03-18:
Also: note that media-stream-library is a peer dependency of media-stream-player, so you'll need to install it explicitly.

ErikMN, 2023-03-20:
I had the same problem, I worked around it by using react-app-rewired described here: https://github.com/AxisCommunications/media-stream-library-js/issues/743

steabert, 2023-03-23:
Duplicate of #743

closed 2023-03-23 by steabert