AxisCommunications/media-stream-library-js #729: CORS origin request blocked when using 'Only HTTPS' on AXIS Camera

AlessandroMartinelli, 2023-02-08:
Hi,
I've been using AXIS media-stream-library for a while, in particular I'm using example [simple.html](https://github.com/AxisCommunications/media-stream-library-js/blob/main/example-streams-web/camera/simple.html) to access AXIS cameras stream from browser. 

Now, we're planning to switch from HTTP to HTTPS for all the communications, including AXIS camera related ones. However I noticed that, if I configure my AXIS M2025-LE Network Camera with "allow access by HTTPS only" (from the web interface) I have the following error when running [simple.html](https://github.com/AxisCommunications/media-stream-library-js/blob/main/example-streams-web/camera/simple.html):
[image]

The error occurrs with Firefox and with Edge as well.

Could you please help me understand how to solve it?

wdichler, 2023-02-09:
You need to change the urls in simple-player.js:
http => https
ws => wss

https://${host}/axis-cgi/usergroup.cgi
wss://${host}/rtsp-over-websocket

AlessandroMartinelli, 2023-02-09:
It worked! thank you very much

closed 2023-02-09 by AlessandroMartinelli