AxisCommunications/media-stream-library-js #743: BasicPlayer in react application with npm package makes error

torcsvarizs, 2023-03-19:
I started a new React app with create-react-app, add to App.js these lines:

```react-js
import { BasicPlayer } from 'media-stream-player'

...

<BasicPlayer
    hostname={window.location.host}
    format="RTP_H264"
    autoPlay
    autoRetry
    vapixParams={{resolution: '800x600'}}
/>
```

I installed package with npm, package json was extended with this deoendency: "media-stream-player": "^12.1.1",

And when I do npm run start I got this message:

```error
Module not found: Error: Can't resolve 'process/browser' in '/Users/zsombortorcsvari/Work/ARH/axis-frontend2/node_modules/media-stream-library/dist'
Did you mean 'browser.js'?
BREAKING CHANGE: The request 'process/browser' failed to resolve only because it was resolved as fully specified
(probably because the origin is strict EcmaScript Module, e. g. a module with javascript mimetype, a '*.mjs' file, or a '*.js' file where the package.json contains '"type": "module"').
The extension in the request is mandatory for it to be fully specified.
Add the extension to the request.
ERROR in ./node_modules/media-stream-library/dist/browser.mjs 84:0-44
Module not found: Error: Can't resolve 'process/browser' in '/Users/zsombortorcsvari/Work/ARH/axis-frontend2/node_modules/media-stream-library/dist'
Did you mean 'browser.js'?
BREAKING CHANGE: The request 'process/browser' failed to resolve only because it was resolved as fully specified
(probably because the origin is strict EcmaScript Module, e. g. a module with javascript mimetype, a '*.mjs' file, or a '*.js' file where the package.json contains '"type": "module"').
The extension in the request is mandatory for it to be fully specified.
Add the extension to the request.

webpack compiled with 1 error
```

In examples in this project get the media-stream-player from the workspace, how can I use it from npm package? What I am doing wrong?

Thank you the answer!
Zsombor

lekoaf, 2023-03-21:
Can verify the error with a clean CRA install.

torcsvarizs-ar, 2023-03-22:
> I have the same issue with create-react-app ("react-scripts": "5.0.1") I have installed media-stream-library, media-stream-player, styled-components and luxon.
> 
> I manage to get it to work with this hack:
> 
>     * Install react-app-rewired: https://www.npmjs.com/package/react-app-rewired
> 
>     * Put config-overrides.js in the project root (same dir as package.json):
> 
> 
> `module.exports = function override(config, env) { //do stuff with the webpack config... config.module.rules.unshift({ test: /\.m?js$/, resolve: { fullySpecified: false, }, }); return config; }`
> 
>     * Wrap all react-scripts calls with react-app-rewired
> 
> 
> Now I get the video player but if I don't put the player in a container that is the same size of the resolution I get this error: `ResizeObserver loop limit exceeded`

Your hack is worked for me also, thanks! But it would be better if it doesn't needed.

And I also get this error: "Now I get the video player but if I don't put the player in a container that is the same size of the resolution I get this error: `ResizeObserver loop limit exceeded`"

@lekoaf is there a solution for this, or we should open a new issue about it?

steabert, 2023-03-23:
> Can verify the error with a clean CRA install.

Can you provide a link to your test repo? Thanks.

steabert, 2023-03-23:
Duplicate of #726

steabert, 2023-03-23:
When you import the library as a module to then bundle it, you need to provide some replacements for native node modules, as `stream`, `buffer`, and `process` dependencies are marked as external. Apparently CRA (or webpack in this case) has an issue with the process dep: https://stackoverflow.com/questions/72410912/webpack-5-breaking-changes-process-browser

Anyway, we could provide a bundled export as well that already includes these dependencies, it just requires some extra configuration.

torcsvarizs, 2023-03-23:
Hello!

Can you just rewrite one of your example using your lib not from your workspace?
If there would an example where the lib comes from the deployed npm package we would be able to see the configuration from there.
I think it would be very useful and make our task easier.

Thank you!

lekoaf, 2023-03-23:
Good idea. We'll take a look at that.

lekoaf, 2023-03-24:
> > Can verify the error with a clean CRA install.
> 
> Can you provide a link to your test repo? Thanks.

https://github.com/lekoaf/msl-test-repo

torcsvarizs-ar, 2023-04-21:
Helo!

Do you have any update for this issue?

Thanks!

steabert, 2023-09-23:
> Helo!
> 
> Do you have any update for this issue?
> 
> Thanks!

I made a PR to fix this problem by pre-bundling some dependencies so that CRA works, which I linked to this issue hoping that would solve your problem. But then I forgot about acutally merging it. It's on its way in now.

PR #749 (2023-09-23): feat: export modules that include dependencies
Some dependencies are difficult to handle by external bundlers, especially the node deps replaced for the browser.

In that case, we can allow to e.g.:

```
import { ... } from 'media-stream-player/heavy'
```

which already includes those dependencies.
Not everything is bundled though.

Fixes #743
closed 2023-09-23 by steabert