AxisCommunications/media-stream-library-js #744: Usage with Angular

tigrenok00, 2023-03-19:
Hi, 
I'm using Angular 15, no webpack.

I've installed `"media-stream-library": "^12.1.1"`
and added to scripts in angular.json:
`"node_modules/media-stream-library/dist/media-stream-library.min.js"`

When I import like this:
`import { Html5CanvasPipeline } from "media-stream-library/dist/media-stream-library.min.js";`

the error is 
`Error: Module not found: Error: Package path ./dist/media-stream-library.min.js is not exported from package C:\XX\XX\XX\node_modules\media-stream-library`

I've looked around at the related issues but no other import worked.
Can you please help?

lekoaf, 2023-03-21:
I'm afraid that me and the rest of the team working on this project know very little about Angular. We mainly work with React.

However, I did manage to find the `Html5CanvasPipeline` in a fresh React app. Maybe you can try that?

[image]

tigrenok00, 2023-03-21:
Hi!
yes - this import works and I see the class!  -however, on build it says:

`Error: Module not found: Error: Package path ./dist/src/index.browser is not exported from package C:\XX\XX\XX\node_modules\media-stream-library (see exports field in C:\XX\XX\XX\node_modules\media-stream-library\package.json)`

steabert, 2023-03-21:
The correct way to use the library if you are importing it as a  module is to use:
```
import { Html5CanvasPipeline } from 'media-stream-library'
```
other import will not work. Also, you will need Typescript >= 4.7 and `moduleResolution: nodenext` for typescript to respect the exports field in package.json (https://www.typescriptlang.org/tsconfig#moduleResolution)
Note that when bundling, you might need to alias `stream` to `stream-browserify` if the bundler does not respect the browser field in package.json.

As interest seems to be growing to use imports and TypeScript, I will try to add more info in the docs how to set this up.

tigrenok00, 2023-03-22:
@steabert Hi I did research migrating to nodenext following your first message, but realized that we couldn't do it because of all the dependencies. The new version imports work! I am going to proceed now with the integration. 
Thanks so much for the fast response!

lekoaf, 2023-03-22:
@tigrenok00 May I ask what you are using this library for? We are curios to know how others use it. :smile:

tigrenok00, 2023-03-22:
@lekoaf Sure! We use it for our clients who have the camera setup and want to integrate the video stream into our web app, also with camera controls. The idea is to control the camera and take images which we will run some image recognition processes on as part of the business logic. We've selected Axis as the camera vendor specifically after researching the different integration capabilities it can support over other options.

lekoaf, 2023-03-22:
Very cool. Thanks! :smiley: