AxisCommunications/media-stream-library-js #748: No streaming in browser

tigrenok00, 2023-03-22:
Hi!

I'm using angular with latest library version. 
I was able to import all the pipelines and when I do play() on the pipeline I see in the Network tab the call to **rtsp-over-websocket (Upgrade: websocket, with status code 101)** which looks exactly the same as the one I see when I go the axis camera public url and watch the stream from there.

However, I am unable to see any streaming in my browser - I based the code on the examples and tried both with video element and canvas, but nothing is shown (the canvas is empty and the video element is unable to play anything).

No errors in the console (I see the debug output I put on "ready", and if I click play again it outputs "incoming stream finished").

Do you have any idea what to check?

This is my code, for both video and canvas options.

TS:

```
export class AdcComponent {
  public vidPipeline: Html5VideoPipeline;
  public canPipeline: Html5CanvasPipeline;

  @ViewChild("canvasEl", { static: false }) canvasEl: ElementRef;
  @ViewChild("videoEl", { static: false }) videoEl: ElementRef;

  encoding: "h264" | "jpeg" = "h264";

  ngAfterViewInit() {
    this.auth();
  }

  public PlayVideo() {
    this.vidPipeline?.close();
    this.canPipeline?.close();

    if (this.encoding == "h264") {
      (this.canvasEl.nativeElement as HTMLCanvasElement).style.display = "none";
      let mediaElement = this.videoEl.nativeElement as HTMLVideoElement;
      this.vidPipeline = new Html5VideoPipeline({
        ws: {
          uri: `ws://XX:XX@XX:8080/rtsp-over-websocket`,
        },
        rtsp: {
          uri: `rtsp://XX:8080/axis-media/media.amp?videocodec=h264`,
        },
        mediaElement,
        auth: {
          username: "XX",
          password: "XX",
        },
      });

      // Restart stream on RTCP BYE (stream ended)
      this.vidPipeline.rtsp.onRtcp = (rtcp) => {
        console.log(rtcp);
        if (isRtcpBye(rtcp)) {
          console.log("bye");
          setTimeout(() => this.PlayVideo(), 0);
        }
      };

      this.vidPipeline.rtsp.onError = (err: RTSPResponseError) => {
        console.log(
          "---pipeline3.rtsp.onError--" + err.code + "--" + err.message
        );
      };

      this.vidPipeline.ready.then(() => {
        console.log("vidPipeline ready");
        this.vidPipeline.play().then(
          () => {
            mediaElement.play();
          },
          (e) => {
            console.log(e);
          }
        );
      });
    } 

  else {
      (this.videoEl.nativeElement as HTMLVideoElement).style.display = "none";
      let mediaElement = this.canvasEl.nativeElement as HTMLCanvasElement;
      console.log(mediaElement);
      this.canPipeline = new Html5CanvasPipeline({
        ws: {
          uri: `ws://XX:XX@XX:8080/rtsp-over-websocket`,
        },
        rtsp: {
          uri: `rtsp://XX:XX@XX:8080/axis-media/media.amp?videocodec=jpeg`,
        },
        mediaElement,
        auth: {
          username: "XX",
          password: "XX",
        },
      });

      // Restart stream on RTCP BYE (stream ended)
      this.canPipeline.rtsp.onRtcp = (rtcp) => {
        console.log(rtcp);
        if (isRtcpBye(rtcp)) {
          setTimeout(() => this.PlayVideo(), 0);
        }
      };

      this.canPipeline.ready.then(() => {
        console.log("canPipeline ready");
        this.canPipeline.play();
      });
    }
  }

  async auth() {
    // Force a login by fetching usergroup
    try {
      await window.fetch(`http://XX:8080/axis-cgi/usergroup.cgi`, {
        credentials: "include",
        headers: {
          "Axis-Orig-Sw": "true",
          "X-Requested-With": "XMLHttpRequest",
        },
        mode: "no-cors",
      });
    } catch (err) {
      console.error(err);
    }
  }
}

```

HTML:

```
<button (click)="PlayVideo()">Play</button>

<div style="width: 500px; height: 500px">
  <video
    onloadedmetadata="this.muted=true"
    #videoEl
    id="videoEl"
    style="width: 100%; height: 100%; object-fit: contain"
    autoplay
    muted
    controls
    oncanplay="this.play()"
  ></video>
  <canvas
    #canvasEl
    id="canvasEl"
    style="width: 100%; height: 100%; object-fit: contain"
  ></canvas>
</div>

```

lekoaf, 2023-03-22:
Do you get the call to the `usergroup.cgi` and the digest login?

tigrenok00, 2023-03-22:
@lekoaf Yes I do, I handle the auth before I press play. And in the network tab I see the digest auth in the request headers to the ws.

tigrenok00, 2023-03-22:
As you can see, I've added the user & pass just about everywhere - both in the URI and the pipeline as well as the call to usergroup.cgi anyway, in hopes we can avoid the browser prompt so the user won't have to enter credentials - don't know if this is possible

tigrenok00, 2023-03-22:
[image]

steabert, 2023-03-23:
> As you can see, I've added the user & pass just about everywhere - both in the URI and the pipeline as well as the call to usergroup.cgi anyway, in hopes we can avoid the browser prompt so the user won't have to enter credentials - don't know if this is possible

You won't be able to avoid a browser prompt connecting directly to the camera, it will always intercept digest auth. The auth component that is provided is mostly for supporting basic auth (mostly not relevant any more), or digest in combination with running a CLI from Node.js (so not in the browser).

You should be able to activate the debug logs by setting `localStorage.debug = 'msl:*'` and reloading.

tigrenok00, 2023-03-23:
@steabert Hi! thank you for the debug logs - it is really helpful! I actually found the problem - I called `pipeline.play()` instead of `pipeline.rtsp.play()` - my mistake. Now the streaming works!

About the auth: we can try to establish some proxy and handle the digest in the backend. We work with AWS, so we can set up a websocket with cognito authorization - in this way how do we pass an auth token or header to the pipeline to use with ws URI? And is rtsp URI needed to be also provided in this case?
Do you happen to have some examples for the backend proxy component?

revanreddy, 2023-06-26:
Any Examples with proxy support to authenticate ws connection to play camera with out prompt ?

Thanks

steabert, 2023-09-23:
> @steabert Hi! thank you for the debug logs - it is really helpful! I actually found the problem - I called `pipeline.play()` instead of `pipeline.rtsp.play()` - my mistake. Now the streaming works!

Great! I'll close this then.

> About the auth: we can try to establish some proxy and handle the digest in the backend. We work with AWS, so we can set up a websocket with cognito authorization - in this way how do we pass an [...]

If I find some time I can try to add some kind of auth proxy as an example.
If you implement this yourself then your cloud service proxy would intercept responses from the camera that as for authentication and then re-send the requests with the proper authentication headers.

closed 2023-09-23 by steabert