AxisCommunications/media-stream-library-js #750: Avoid Prompt for camera authentication

revanreddy, 2023-04-05:
When We load the axis camera on browser it prompts to enter the credentials to stream. but we want to avoid the prompt and authenticate by passing the credentials through request.

Thanks

MennoSaey, 2023-05-23:
Have you solved your problem? i am dealing with the same issue.

revanreddy, 2023-07-07:
@MennoSaey  No didn't get any.. did  you find any way ?

MennoSaey, 2023-07-07:
@revanreddy Sadly i didn't find a solution either. The way we tried to solve it was by sending a simple authenticated request but even when providing the correft headers there was still an issue, I have spoken to someone associated with Axis and they confirmed it is an issue on their end and will work on fixing this.

revanreddy, 2023-07-07:
Thanks @MennoSaey for the quick response..

MennoSaey, 2023-07-07:
@revanreddy No, our implementation only needed to view the cameras. I forgot to mention that if u just want to view the camerafeed, you can send a post request that changes the anonymous viewing to enabled. This way you can view the camerafeed and then disable it again after you are done.

revanreddy, 2023-07-07:
Thanks @MennoSaey. It will be helpful for us :)

steabert, 2023-09-24:
The auth module we have was mostly used for testing purposes and I haven't kept it up-to-date with regards to docs. The digest auth part of it only works in Node.js, if you run it from the command line.

When running in a browser, you cannot intercept digest auth. The only way to do that is to run a proxy that can intercept any digest auth request and provide authentication headers. But that is slightly out-of-scope for this project. I'll have a look if I can find some good example(s) on how to do this. However, note that you then no longer just access the camera directly from the browser.

I have not tested with cameras with basic auth, is that what your camera is set to? In theory, that should work with the provided auth module.

SampsaKaskela, 2023-11-30:
You can accomplish this by giving Authorization header in your request. For example when using usergroup.cgi. In this case CORS has to be configured since you can't use no-cors mode like in examples. I have following function that works for me and it uses Basic auth.

```
const authorize = async (hostname: string) => {
  const response = await window.fetch(`http://${hostname}/axis-cgi/usergroup.cgi`, {
    credentials: 'include',
    headers: {
      Authorization: `Basic ${window.btoa('username:password')}`,
    },
  });
  if (!response.ok) throw new Error('Failed to get stream');
};
```

closed 2025-01-12 by steabert