AxisCommunications/media-stream-library-js #752: Display Issues with HIKVISION DS-2CD6D24FWD-IZHS in the form of green artifacts

Seliah, 2023-06-08:
Hi guys! First of all, thank you so much for maintaining and providing this awesome library.

I am currently encountering a problem with display issues that seem like artifacts in RTSP streaming. I would be very happy to know what you guys think might cause effects like this. Also I would be very interested in the steps I need to take or functions I need to implement to fix this. 

**Description**
Artifacts like in the screenshots below pop up in a high resolution (Full HD) RTSP stream with a specific camera type. This happens after starting the stream and stays for a few seconds. It does not always happen but most of the times. The camera type is the DS-2CD6D24FWD-IZHS by HIKVISION. This is a device with 4 cameras which will be identified as one device with many profiles in ONVIF. You can find it here: https://www.hikvision.com/en/products/IP-Products/Network-Cameras/Panoramic-Series/DS-2CD6D24FWD--I--Z--H--S---NFC-/.

In native players like VLC or ffplay, this problem does not occur - even when opening the stream around 10 times simultaneously.

Reducing the bitrate of the streams fixes the artifacts. Using a minimal bitrate is not optimal in my use case though.

I inserted the debug logs of this library that occur in the described event below. I noticed that the following messages will be repeated as long as the stream stays up:
```
16:13:42.642 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 7 +2s
16:13:42.643 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 8 +1ms
16:13:42.643 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 6 +0ms
```

NAL types 6, 7 and 8 are called SEI, SPS and PPS. I was wondering if the correct interpretation of these packets would solve my problem. Do you think that could be a possibility?

**To Reproduce**
Reproducing is likely very hard as this problem only pops up in the case of this exact camera type. It happens on high bitrate Full HD Streams, when opening them up.

**Screenshots**
Green overlaying artifacts:
[image]

Occasional mild artifacts:
[image]

**Environment:**
 - OS: Linux - Pop_OS!
 - Browsers: Chrome, Firefox, Opera, Vivaldi
 - Version: 12.2.0

**Steps I took**
- I updated this library to the newest version, tried old versions like 6.0.4 as well
- Minimized the usage of the camera to sort out problems of parallel nature
- I analyzed the logs for errors or warnings

**Logs**
```
16:13:42.528 browser.mjs:268 msl:rtsp:outgoing DESCRIBE rtsp://192.168.151.73:554/Streaming/Channels/401?transportmode=unicast&profile=Profile_401 RTSP/1.0
CSeq: 2
Accept: application/sdp

 +0ms
16:13:42.535 browser.mjs:268 msl:rtsp:outgoing DESCRIBE rtsp://192.168.151.73:554/Streaming/Channels/401?transportmode=unicast&profile=Profile_401 RTSP/1.0
CSeq: 2
Accept: application/sdp
Authorization: <removed for security>

 +0ms
16:13:42.543 browser.mjs:268 msl:rtsp:incoming RTSP/1.0 200 OK
CSeq: 2
Content-Type: application/sdp
Content-Base: rtsp://192.168.151.73:554/Streaming/Channels/401/
Content-Length: 693

v=0
o=- 1685981622039637 1685981622039637 IN IP4 192.168.151.36
s=Media Presentation
e=NONE
b=AS:5050
t=0 0
a=control:rtsp://192.168.151.73:554/Streaming/Channels/401/?transportmode=unicast&profile=Profile_401
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:5000
a=recvonly
[... 22 lines cut ...]
 +0ms
16:13:42.558 browser.mjs:268 msl:rtsp:outgoing PLAY rtsp://192.168.151.73:554/Streaming/Channels/401/?transportmode=unicast&profile=Profile_401 RTSP/1.0
CSeq: 4
Range: npt=0-
Session: 928271617
Authorization: <removed for security>

 +0ms
16:13:42.602 browser.mjs:268 msl:rtsp:incoming RTSP/1.0 200 OK
CSeq: 4
Session:        928271617
RTP-Info: url=rtsp://192.168.151.73:554/Streaming/Channels/401/trackID=1?transportmode=unicast&profile=Profile_401;seq=20197;rtptime=1375439940
Date:  Mon, Jun 05 2023 16:13:42 GMT

 +0ms
16:13:42.642 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 7 +2s
16:13:42.643 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 8 +1ms
16:13:42.643 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 6 +0ms
16:13:42.724 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 7 +81ms
16:13:42.725 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 8 +0ms
16:13:42.725 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 6 +1ms
16:13:42.915 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 7 +190ms
16:13:42.916 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 8 +1ms
16:13:42.917 browser.mjs:268 msl:h264depay H264depayComponent can only extract types 1,5 and 28, got 6 +1ms
```

steabert, 2023-09-23:
The NAL units that are `SPS` and `PPS` are probably OK to ignore, we cannot handle these in-band but instead we rely on this information to be sent out-of-band in the `sprop-parameter-sets` in the SDP (which according to your logs seems to be there).

However, not parsing the `SEI` NAL unit might be the problem. If you can find out more about what this does you could potentially add support for these?

Seliah, 2023-09-25:
Thank you for this information! We are not sure about adding in additional support yet but we will let you guys know.

closed 2024-07-12 by steabert