AxisCommunications/media-stream-library-js #754: "process.env" gets set to empty object in Vite dev environment when importing media-stream-player

SampsaKaskela, 2023-08-25:
**Describe the bug**
When importing from media-stream-player (in my case BasicPlayer) in Vite the process.env gets reset to empty object. Removing any imports to media-stream-player makes the process.env to have the defined values.

**To Reproduce**
In Vite config use "define" field to define process.env values for example
```
defineConfig({
    plugins: [react()],
    define: {
      'process.env.MY_ENV': JSON.stringify('my_env_value'),
    }
});
``` 
In your code, import BasicPlayer (or any component from media-stream-player) and use it in React component. If you now try to console.log(process.env) it is completely empty instead of having the defined values. This is only problem when running in dev server since in production Vite replaces process.env mentions. So in production everything works.

**Environment (please complete the following information):**
 - Windows 10 Enterprise
 - Chrome 116.0.5845.111 (Official Build) (64-bit), Nodejs 18.16.0
 - media-stream-player 12.2.0
 - Vite 4.4.9

**Additional context**
After debugging it seems that problem occurs when media-stream-library package imports process/browser and this loads process polyfills which sets the process.env to empty object. I tweaked Vite config and following seems to fix the problems

```
defineConfig({
    plugins: [react()],
    define: {
       global: 'window',
       'process.env.MY_ENV': JSON.stringify('my_env_value'),
    },
    resolve: {
        alias: {
          'process/browser': 'process/index.js'
        },
      },
    }),
});
``` 
Defining global is needed because global doesn't exist otherwise. Alias for process/browser is used to force it to load the global.process (which due to global definition is actually window.process) instead of the polyfill that causes process.env to be set to empty object. This makes the dev environment work but this then in return potentially breaks the build if some libraries rely for example on global definition. So to completely fix this these tweaks need to be made conditional to dev environment. So eg.
```
defineConfig(({ mode }) => ({
  plugins: [react()],
  define: {
    'process.env.MY_ENV': JSON.stringify('my_env_value'),
    ...(mode === 'development' && {
      global: 'window',
    }),
  },
  ...(mode === 'development' && {
    resolve: {
      alias: {
        'process/browser': 'process/index.js',
      },
    },
  }),
}));
```
Far from ideal so hoping this could be fixed so this workaround would not be needed.

steabert, 2023-09-23:
Thanks for your report. I'm not an export on all the polyfill stuff, but would something like this work in `polyfill.mjs`:

```ts
if (window.process === undefined) {
  window.process = process
}
```

SampsaKaskela, 2023-09-23:
This would 

> Thanks for your report. I'm not an expert on all the polyfill stuff, but would something like this work in `polyfill.mjs`:
> 
> ```ts
> if (window.process === undefined) {
>   window.process = process
> }
> ```

This would not work since process in not undefined since it is polyfilled. Also the moment process gets polyfilled and process.env gets set to empty then all the process.env data gets lost in that moment. Essentially there should be way to polyfill it and then fill the process.env some way afterwards with the data that Vite passes in or avoid polyfilling parts that already have been defined. I guess this same issue was also present in Create React App project.

steabert, 2023-09-24:
I don't think I understand what you're saying. The solution I presented would not polyfill `process` if it already exists?

steabert, 2023-09-24:
Looking at `process/browser` it's indeed doing that. I'm not sure why I went with this solution though, it should be enough to remove it altogether from the bundle.

closed 2023-09-24 by steabert

steabert, 2023-09-24:
This should be solved as of version `13.0.0` but I would appreciate if you can test that it actually does 😄

SampsaKaskela, 2023-09-24:
Ok I will give this a try to see if it works and provide reproduction repo if issue persists.

SampsaKaskela, 2023-09-25:
I have confirmed that it seems to work now without any additional config required in Vite.

commit f05f40b5 (2023-09-24): fix: custom polyfill process/browser

Always bundle `process/browser` and rename the global variable to
`process_browser`.

This should selectively polyfill it only for the `readable-stream`
dependency which we use, and not interfere with any other usage
of `process`.

Fixes #754
commit 0f71052b (2023-09-24): fix!: bundle stream-browserify/process by default

For browsers, always bundle `stream` from `stream-browserify`,
`buffer` and `process`. Include polyfill for `process/browser`
using the custom global variable name `process_browser` and
set a global `Buffer`.

This should selectively polyfill it only for the `readable-stream`
dependency which we use, and not interfere with any other usage
of `process`.

BREAKING CHANGES:
 - The `heavy` variants are no longer in use, this is now the default.
   Replace `media-stream-library/heavy` with `media-stream-library`.
   Replace `media-stream-player/hea [... cut]
commit bf5e114d (2023-09-24): fix!: bundle stream-browserify/process by default

For browsers, always bundle `stream` from `stream-browserify`,
`buffer` and `process`. Include polyfill for `process/browser`
using the custom global variable name `process_browser` and
set a global `Buffer`.

This should selectively polyfill it only for the `readable-stream`
dependency which we use, and not interfere with any other usage
of `process`.

BREAKING CHANGES:
 - The `heavy` variants are no longer in use, this is now the default.
   Replace `media-stream-library/heavy` with `media-stream-library`.
   Replace `media-stream-player/hea [... cut]
commit cf439574 (2023-09-24): fix!: bundle stream-browserify/process by default

For browsers, always bundle `stream` from `stream-browserify`,
`buffer` and `process`. Include polyfill for `process/browser`
using the custom global variable name `process_browser` and
set a global `Buffer`.

This should selectively polyfill it only for the `readable-stream`
dependency which we use, and not interfere with any other usage
of `process`.

BREAKING CHANGES:
 - The `heavy` variants are no longer in use, this is now the default.
   Replace `media-stream-library/heavy` with `media-stream-library`.
   Replace `media-stream-player/hea [... cut]