AxisCommunications/media-stream-library-js #990: Cannot use versions >v11.2.0 in a pure ESM project

vetlevo, 2024-07-11:
Hello! We're using the media-stream-library in a NodeJS project that is configured to only allow 'import' syntax. This still allows importing CommonJS modules, and we haven't had any issues with this in the past. 

However, if we upgrade the media-stream-library beyond v11.2.0, we get the following error at runtime (This is with v13.1.1):
```
node_modules/media-stream-library/dist/node.mjs:12
  throw Error('Dynamic require of "' + x + '" is not supported');
        ^
Error: Dynamic require of "tty" is not supported
    at file:/node_modules/media-stream-library/dist/node.mjs:12:9
    at ../node_modules/debug/src/node.js (file:/node_modules/media-stream-library/dist/node.mjs:1129:15)
    at __require2 (file:/node_modules/media-stream-library/dist/node.mjs:15:50)
    at ../node_modules/debug/src/index.js (file:/node_modules/media-stream-library/dist/node.mjs:1306:24)
    at __require2 (file:/node_modules/media-stream-library/dist/node.mjs:15:50)
    at file:/node_modules/media-stream-library/dist/node.mjs:2020:28
    at ModuleJob.run (node:internal/modules/esm/module_job:218:25)
    at async ModuleLoader.import (node:internal/modules/esm/loader:329:24)
    at async loadESM (node:internal/process/esm_loader:28:7)
    at async handleMainPromise (node:internal/modules/run_main:113:12)

Node.js v20.11.0
```

From what I can tell, there seems to be require syntax in an mjs file that causes the issue.

Any thoughts?

vetlevo, 2024-07-11:
For now, our fix is to remove the constraints in our project to be an ES module. This works for us as we currently don't have other dependencies in this project that have the same constraints.

steabert, 2024-07-12:
Yes, this is a head-ache for us too, I'll have to have another look at this. I think it might be difficult since we ourselves depend on things that only are CommonJS.

steabert, 2024-07-12:
It looks like it's the debug package, we should probably replace it with a simple utility or something like `tslog`, because from the comments on their repo it doesn't seem like they will give up CommonJS any time soon.

jessypouliot98, 2024-12-16:
For anyone having an issue using Next.js when running `next build` a simple solution for us was to wrap the React component that was making use of this library (media-stream-library) in a dynamic import.

```tsx
import dynamic from "next/dynamic";
// import { MediaStreamLibComponent } from "./";

const MediaStreamLibComponent = dynamic(() => (await import("./")).MediaStreamLibComponent, { ssr: false });

export default const Page = () => {
  return (
    <MediaStreamLibComponent/>
  );
}
```

commit 21fc897c (2025-01-08): feat(streams): use Web Streams API

Replace Node.js stream module with Web Streams API.
Since the latter is substantially different in some important details
regarding stream pipelining, the way pipelines are built are redefined.
The component concept as module blocks is removed and instead the
components themselves expose streams that can be combined together.
The pipelines are then simple stream compositions.

Because this is a major (breaking) change, it's done in concert with
other planned improvements that are also breaking changes (see section
below for details).

Improvements:
- replaci [... cut]
commit d70605bf (2025-01-08): feat(streams)!: use Web Streams API

Replace Node.js stream module with Web Streams API.
Since the latter is substantially different in some important details
regarding stream pipelining, the way pipelines are built are redefined.
The component concept as module blocks is removed and instead the
components themselves expose streams that can be combined together.
The pipelines are then simple stream compositions.

Because this is a major (breaking) change, it's done in concert with
other planned improvements that are also breaking changes (see section
below for details).

Improvements:
- replac [... cut]
commit 0fa7fcc5 (2025-01-08): feat(streams)!: use Web Streams API

Replace Node.js stream module with Web Streams API.
Since the latter is substantially different in some important details
regarding stream pipelining, the way pipelines are built are redefined.
The component concept as module blocks is removed and instead the
components themselves expose streams that can be combined together.
The pipelines are then simple stream compositions.

Because this is a major (breaking) change, it's done in concert with
other planned improvements that are also breaking changes (see section
below for details).

Improvements:
- replac [... cut]
commit a660c8bf (2025-01-08): feat(streams)!: use Web Streams API

Replace Node.js stream module with Web Streams API.
Since the latter is substantially different in some important details
regarding stream pipelining, the way pipelines are built are redefined.
The component concept as module blocks is removed and instead the
components themselves expose streams that can be combined together.
The pipelines are then simple stream compositions.

Because this is a major (breaking) change, it's done in concert with
other planned improvements that are also breaking changes (see section
below for details).

Improvements:
- replac [... cut]
closed 2025-01-10 by steabert