AxisCommunications/media-stream-library-js #1062: Returning reason on socket close

roycw, 2024-11-25:
Guys we have this code in the socket

socket.onclose = (e) => {
      debug('msl:websocket:close')(`${e.code}`)
      if (e.code === CLOSE_GOING_AWAY) {
        this.onServerClose && this.onServerClose()
      }
      // Terminate the streams.
      incoming.push(null)
      outgoing.end()
    }

is there anyway we can pass the entire e object back to the onServerClose so we can send custom reasons?

steabert, 2025-01-12:
The new major update has some changes, one is that the socket has been moved more out of the actual component, which now only connects the socket to a stream. To listen to server close, you can add your own listener to the socket.

closed 2025-01-12 by steabert

roycw, 2025-01-12:
Wonderful!

Weather can we see list of changes.

CAn u guidee me just for this section of the code on how i can write my own
handler