AxisCommunications/media-stream-library-js #1077: RtspSession.update does not update _sessionControlURL

selimb, 2025-01-03:
**Describe the bug**

I am trying to seek to different times without tearing down and setting up RTSP sessions, and instead reuse the existing one. In theory the RTSP protocol should allow this (as far as I know), and I can *almost* do it with this library. The problem is that calling `.update` on `RtspSession` does not update `_sessionControlURL`, which is what is used to send SDP commands.

**To Reproduce**

1. Apply the following git patch on this repo:
```diff
--- a/streams/tests/rtsp-session.test.ts
+++ b/streams/tests/rtsp-session.test.ts
@@ -78,10 +78,12 @@ describe('rtsp-session send method', (test) => {
 
   test('should use the supplied URI', async (ctx) => {
     const uri = 'rtsp://whatever/path'
+    const uri2 = 'rtsp://new/path'
     const s = new RtspSession({ uri })
+    s.update(uri2)
     const done = new Promise((resolve) => (ctx.resolve = resolve))
     s.outgoing.once('data', (req) => {
-      assert.is(req.uri, uri)
+      assert.is(req.uri, uri2)
       ctx.resolve()
     })
     s.send({ method: RTSP_METHOD.DESCRIBE })
```
2. Run `just test`

This outputs:
```
   FAIL  rtsp-session send method  "should use the supplied URI"
    Expected values to be strictly equal:  (is)

        ++rtsp://new/path         (Expected)
        --rtsp://whatever/path    (Actual)
```

**Environment:**
 - OS: Linux
 - Browser: Firefox
 - Version: 13.1.1

**Possible solution**

I was able to reuse an RTSP session by doing something like this in my code:
```ts
// this._pipeline is a `Html5VideoPipeline`
this._pipeline.rtsp.stop();
const uriNew = ...;
this._pipeline.rtsp.update(uriNew);
this._pipeline.rtsp._sessionControlURL = uriNew;
this._pipeline.rtsp.play();
```

I'm not 100% sure what `.update` *should* be used for since it's not used anywhere in this repo, but AFAIK this can be fixed by simply adding the following at the end of the `.update` implementation:
```ts
    this._sessionControlURL = this.uri;
```
It's probably only safe to do this if the `_state = STATE.IDLE` though? 

I'm happy to work on a PR for this.

steabert, 2025-01-07:
Note: there are some up-and-coming major changes to this, so I suggest you have a look at the `web-streams` branch if this is still an issue!

selimb, 2025-01-07:
> (usually we don't reuse a pipeline)

Is there any reason why? As far as I can tell it should lead to a much snappier experience, no?

steabert, 2025-01-11:
> I'll have a look at the `web-streams` branch! Thanks.

I've added a feature/fix for this in the linked PR, would that work for you?

> > (usually we don't reuse a pipeline)
> 
> Is there any reason why? As far as I can tell it should lead to a much snappier experience, no?

When setting up a new session with a new URI is expensive already, so just starting a new pipeline isn't really problematic (that is why the URI is set in the constructor). The fix I added allows you to override that URI later on, and only treats the original as the default. However, you stated at the beginning that you want to seek to different times, you shouldn't need to change the URI?

Note that `play` doesn't allow you to pick a particular stream (it always streams the whole session).

commit 1746030b (2025-01-13): feat(streams): allow changing the RTSP URI

Adds a URI option to the describe method in the RTSP session.
If provided, it's used for that session instead of the URI that
was provided to the constructor. The latter now only serves as
default URI when none is provided to describe.
On teardown, the session control URL is restored to the default.

Fixes #1077
commit 087bbec4 (2025-04-13): feat(streams): allow changing the RTSP URI

Adds a URI option to the describe method in the RTSP session.
If provided, it's used for that session instead of the URI that
was provided to the constructor. The latter now only serves as
default URI when none is provided to describe.
On teardown, the session control URL is restored to the default.

Fixes #1077
PR #1083 (2025-04-13): feat(streams): allow changing the RTSP URI
Adds a URI option to the describe method in the RTSP session. If provided, it's used for that session instead of the URI that was provided to the constructor. The latter now only serves as default URI when none is provided to describe.
On teardown, the session control URL is restored to the default.

Fixes #1077

**Checklist for review**

- PR title and description are clear
- change follows existing coding style and architecture
- necessary unit tests were added
- documentation was updated
commit 23a6c5d6 (2025-04-13): feat(streams): allow changing the RTSP URI (#1083)

Adds a URI option to the describe method in the RTSP session.
If provided, it's used for that session instead of the URI that
was provided to the constructor. The latter now only serves as
default URI when none is provided to describe.
On teardown, the session control URL is restored to the default.

Fixes #1077
closed 2025-04-13 by steabert